miicam/MiiCam #1: IR_Led not automatically switching on in nightmode

epalzeolithe, 2019-02-03:
Hi Fliphess, you really do a great job.

The last release on this repo is good, no ir_led nvram error in settings.html, and the switch in settings.html.

I'm tracking another issue, the automatic nightmode works well with, but the IR_led or IR_cut does not switch as needed - the picture is totally dark. Manually, using the settings.html, i can enable.

I'm looking into the code to find the issue

fliphess, 2019-02-04:
Hey @epalzeolithe! 

The new auto night mode switcher is not the same tool as the previous one and some manual configuration is required to get it to work properly

To get it to switch at a specific value, you need to find the appropriate IR and EV values and configure the command line arguments accordingly. 

This is currently a manual job as I didn't find the time to find good default values that work similar to the last one. 

Using the verbose flag (`-v`), and while running the service in the foreground (without using the init script), you can monitor the `IR` and `EV` values of the sensor in your terminal.  

The mechanism has been rewritten from scratch without the previous examples, so I'm sure they both work differently.  

I can't use the same default values because I don't have the source code of the old one (`ir_sample`).

fliphess, 2019-02-04:
About the IR Cut mode: This mode is currently not being switched in the mode switcher. Did the previous one switch IRCut? 

I never used the old one because it was blocking the `IR led` while it was enabled.

(But I don't think it's hard to add to the code.)

fliphess, 2019-02-04:
As far as I can see it's not very useful to switch the IR cut setting as it is sort of overrules by night mode (b/w).

It might work if you enable IR cut when switching off the nightmode, but I'm not sure if thats needed. If the IR cut value remains the same after switching off nightmode I think you can just leave it on. 

If IR cut should be switched when a certain value is reached that is possible, but I have no idea at which value of which measurement it should be enabled..... 

Maybe when the IR value is higher than N turn on the IR cut, if IR value is below N switch it off?

fliphess, 2019-02-04:
I've updated [the auto nightmode docs](https://miicam.github.io/Using-Auto-Night-Mode/)

V4n1X, 2019-02-10:
The problem is I see, if its going darker the IR value is going higher, not lower so Dark means {"ev":17,"ir":189}

and bright means:

{"ev":3380,"ir":106}

So it must be:

When the IR value is over 100 the IR led is switched ON
When the IR value is below 100 the IR led is switch OFF

Can you change this?

fliphess, 2019-02-11:
@V4n1X Thanks for your feedback! That sounds like a very plausible solution :) I will have a look at it later this week 👍

fliphess, 2019-02-14:
I've created a fix, which will be released soon. As I put my commit, the issue will be closed, but don't hesitate to reopen it if the behaviour is not as expected/ as it should be. 

Otherwise we should have a look how [other solutions solve this problem](https://github.com/Dafang-Hacks/Main/blob/master/autonight/autonight.c)

closed 2019-02-14 by fliphess

fliphess, 2019-02-22:
Hey @V4n1X! 

I've released the changes on the auto night mode switcher and I've created nightly builds for quick testing. Let me know if the switcher suffices or if more changes are required to make it more useful 👍

commit ec4a1bae (2019-02-14): Change the logic in auto_night_mode as requested (Fixes: #1)