miicam/MiiCam #3: Config Editor in Web Interface not working

V4n1X, 2019-02-10:
Hello,

in the latest Version, if I want to change something in the Config over the WebUI, i get when I Save the Config this ERROR:

Invalid input in configuration content: ROOT_PASSWORD='MyPassword123' (line: 28)

Happens even with the default password, also if I change nothing and want to save the config again, this happens.

fliphess, 2019-02-11:
Hey @V4n1X!

Thanks for reporting this. 

I think it has something to do with the regex that is checked for invalid chars. 
I will have a look at it.

fliphess, 2019-02-11:
Ah! it's [in the regex](https://github.com/miicam/MiiCamWeb/blob/master/libs/system.php#L108)! 

As a workaround, try using double quotes rather than single, that should work until I've found a way to fix it :)

fliphess, 2019-02-11:
/move to web

closed 2019-02-11 by move[bot]

fliphess, 2019-02-22:
This should now be live in the next release. Thanks for letting me know! 👍