miicam/MiiCam #4: The blue LED is always on, and can't be turned off

pcmester, 2019-02-11:
The blue LED is always on, and this is very annoying especially at night. I tried to turn off via the webUI and also with MQTT, but it was not successfull. I haven't see any error message, if I try to turn off, it seems like its turning off, but not or sometimes the LED is turning off and back on again. I tried to wipe the SD card and reconfigure everything, but it was not solve the issue.

V4n1X, 2019-02-11:
Tested it, problem exists and I can reproduce it.

It seems like, its trying to turn it of, but it is switched instantly on, because the value got overwritten I think.

fliphess, 2019-02-11:
Hmz strange.... I can't reproduce it.... 

It could be a browser/webui issue.... Let's verify this first as it's easy to debug: 

- Can you turn on/off the leds on the command line using `blue_led on` and `blue_led off` ? 
- If you open the browser and check if the http requests are being made? There should be 2 requests to enable/disable the leds: `/api/led/blue_led/set/off` and `/api/led/blue_led/set/on`. 

If you visit those urls on your camera in the browser it should show something like:
```json
{"led":"blue_led","action":"TurnOff","success":true}
```

fliphess, 2019-02-11:
I'm thinking of some kind of race condition in the state of the led where the state is set to off while the leds are still on. Let me know if it works by directly opening the on/off http endpoints in your browser. 

If that's not it I have to do some deep diving in the bootup timing :)

V4n1X, 2019-02-11:
So I tried:
* /api/led/blue_led/set/off
{"led":"blue_led","action":"TurnOff","success":false}

then
* /api/led/blue_led/set/on
{"led":"blue_led","action":"TurnOn","success":true}

and then
* /api/led/blue_led/set/off
{"led":"blue_led","action":"TurnOff","success":true}

but nothing changes.

Blue LED is blinking all the time, its not a constant flashing.

Same with the scripts

V4n1X, 2019-02-11:
Oh I see one thing at the moment.

If no movement is detected the blue light is on, no blinking etc..., if movement is detected the blue is blinking for round about 5 seconds.

So it must be to do with movement detection.

pcmester, 2019-02-11:
I tried via SSH, and I got the following error:

```
Failed to access /sys/class/gpio/gpio14
IRCUT initialization failed
```

pcmester, 2019-02-11:
I think, that this is some GPIO issue, because I can't control also the ir_cut in SSH
```
root@camera /root # ir_cut -e
Failed to open gpio value for writing!
*** Error: Failed write to gpio pins!
```

fliphess, 2019-02-11:
@V4n1X  That makes sense :) I'll create some configuration to ask whether the led should blink :)

@pcmester Is the error still there if you reboot the camera? I've seen this error as well, but I thought I fixed it in the last release

pcmester, 2019-02-12:
I've rebooted, and now it is blinking even if there isn't any motion. In SSH, if I try to turn off the LED, first time I haven't got any error (but the LED is on), and neither the second time, but the third time is interesting. There is an another error message:
```
root@camera /root # blue_led off
root@camera /root # blue_led off
root@camera /root # blue_led off
Failed to open /dev/ftpwmtmr010
PWM initialization failed
```

fliphess, 2019-02-12:
@pcmester Can you first turn off motion detection, just to ensure the led is not triggered by motion and after that reboot the camera? 

I think the pwm is locked by the `motion_on.sh` or `motion_off.sh`scripts while in the same time the led is controled through web and or command line, which can cause some of this behaviour.....

pcmester, 2019-02-12:
I turned off motion detection, and it seems like that this is solved the problem, but I can't enable IR_CUT now.
`
root@camera /root # ir_cut -e
root@camera /root # ir_cut -s
*** IR Cut is off
`

fliphess, 2019-02-13:
The IR Cut is indeed a bit flaky at the moment, I'm working on it this week and It will be fixed in the new release next week. But lets focus on the leds in this issue, i'll create a new bug for the IR cut:

If the motion detection is off, do the blue and yellow led function as expected? I'll remove the led indicator for motion in the next release as I've to find a better mechanism for it.

pcmester, 2019-02-13:
Yes, without motion detection turned on, the blue and yellow LEDS functions as normal

fliphess, 2019-02-14:
As a workaround: If you don't want the motion detectionto control the leds, edit the `motion_on.sh` and `motion_off.sh` scripts in `/tmp/sd/firmware/scripts/` and remove the led actions.

pcmester, 2019-02-14:
Thank you, i will do that

closed 2019-02-18 by fliphess

pcmester, 2019-02-20:
I can confirm, that this is solved the issue. Thank you again!

fliphess, 2019-02-21:
@pcmester Thanks! I'm a bit late with the new release due to some things in the non-digital world that needed attention, but I aim at coming weekend :)

fliphess, 2019-02-22:
@pcmester @V4n1X A new release is created a few minutes ago. It should go through the build process on [travis](https://travis-ci.org/miicam/MiiCam/builds/497206274) and it'll be available in an hour ;)