miicam/MiiCam #9: Update over the air (OTA) support would be great

pcmester, 2019-02-25:
Hey @fliphess!
The update process is can be really hard, especially if you mount a camera to the ceiling. Now, I can't update the camera over the air correctly, but i think it is possible. The problem is, that if I try to replace the files that are running, this causes problems and this way it's not great. I know that there are more important features that would be great, this is just a far future feature request.

fliphess, 2019-02-26:
Hey @pcmester! Thanks for this feature request :)

I've been working on such a thing for the last few months.... It's just nog finished yet. 
For the web interface I've come to the point that it's now possible to update on the command line, but I'm keeping it "quiet" for a while until I've done some decent testing. 

For the rest of the hack I'm still working out how to proceed, as it's very easy to brick a camera by updating to the latest version over the air. 

Other firmware hacks solved this issue by including the binaries in the repository, making it possible to simply wget every file to verify if it's changed, but I chose not to take that path to prevent the git repository from becoming slow due to rapid changing binary blobs in my git repository.

Downloading the `tgz` file all at once and unpack it on the camera made me run into memory issues  usage during the download of the latest version, causing the camera to reboot before the download is fully done. 

I think it's all not that hard to overcome so I keep working on it, but don't  expect it before summer. 

In the meantime: rsync is available on the camera, so updating from your desktop should not be that hard if you stop all services first and after that rsync all the changes to the camera:

```
rm sdcard/config.cfg
rsync -avz -e 'ssh -l root' sdcard/* /tmp/sd
ssh root@camera "/tmp/sd/firmware/scripts/update/configupdate /tmp/sd/config.cfg"
```
(This is all untested, so be careful and test your commands carefully with a single file prior to syncing all files.)

fliphess, 2019-02-26:
If you manage to get the remote update to work, feel welcome to add a PR for an `update_camera.sh` remote update script in `tools/dev` that downloads the latest release on your laptop, uses rsync to copy the content of the `tgz` file to the camera and updates the configuration file afterwards as shown in my earlier post.

fliphess, 2019-05-02:
@pcmester, Sorry for the late response and the annoying bot.....

Did you checkout the updater? (https://github.com/miicam/MiiCamUpdater) 
It's the steps I mentioned in a script, and I'm open to suggestions for improvement. 

It's not a very shiny updater, but I gave up on updating from the camera itself as it kept crashing, dropping packets, saving corrupt downloads and all the nasty stuff you've to go through when designing an OTA mechanism for this type of hardware. I might be able to port an existing linux package manager to this camera and create an updater utility, but for now this is a small utility to help you update. :)

pcmester, 2019-05-04:
Not a problem. Now I can access phisically the camera much easier than before, so I can update manually if it necesarry. The latest version did not want to start, now I use the 2nd release before the latest, I do not know what is the problem, but it just not boots to the hack with the latest. I will check out the updater, if new release comes out, I will try to update to that. Thanks for your reply, and the developement for this camera.

closed 2019-07-06 by fliphess