miicam/MiiCam #10: Color problems - the picture is pink (or someting else color, but not realistic)

pcmester, 2019-02-27:
Hey @fliphess!

I updated to the latest release about 1,5 day ago, and there is an annoying issue about colours. After i updated, the whole picture turned to pink colored. I hoped, that this is caused by an incorrect setting, so i pressed the reset button on the camera settings page, and nothing changed. Today I tried to adjust the hue, but if i do that, only the "filter colour" changes. The colours are not realistic

roboterm, 2019-02-27:
I think this is caused by IR_Cut.

fliphess, 2019-02-27:
The reset button loads the default state, which is IR_Cut OFF. 

Can you check what happens when you enable IR_Cut? 
It sounds like the filter is OFF while there is a lot of IR led, resulting in a pink frame. (more red light than blue)

V4n1X, 2019-02-27:
Yes this is the IR_Cut, i am working on optimal variables for the auto night mode, but I dont know if the IR Cut is useful in any way, if IR-Cut is of, i only need the night mode + Irled for dark and both off for day mode. So is it possible to disable the switch of the ir cut?

roboterm, 2019-02-27:
I don’t understand the ir_cut functionality either. 
Can’t we just link the ir_led to the nightmode?
If nightmode is set on ir_led does as well and vice versa.

V4n1X, 2019-02-27:
Yes that is a good deal I think, Nightmode on means IR Led on, nightmode of, also Ir led off.

IRCut should be globally disabled, its useless I think.

Then we can adjust the auto night mode feature better.

fliphess, 2019-02-27:
@V4n1X  In the afternoon when there is a lot of red light, and the sun is in a low position, I couldn't do without IR  Led to correct the red from the image (My main window is on the south side). 

I'm reluctant to disable IR cut functionality while auto_night_mode is on, as the nightmode switcher responds to low light, where the IR cut functionality is meant for use when there is too much light.

fliphess, 2019-02-27:
I think it's time to rename the `auto_night_mode` utility, as it's not just nightmode (black and white) anymore but ir_cut and ir_led as well.... :) 

At startup I need to switch the `IR Cut` and `IR Led` at least once to init the `GPIO` pins, so globally disabling would not be an option as control would be disabled with state ON...... 

(which implies you cannot control it, but it's still functional and active)

fliphess, 2019-02-27:
@V4n1X I'm closing this issue, as it's about a pink picture and we're drifting off to the nightmode switcher :) 

I'll create a new issue regarding the auto_night_mode switcher utility, please add any suggestion, comments, ideas and feature request about the switcher in (#12), so we can do some improvement :)

closed 2019-02-27 by fliphess

pcmester, 2019-02-27:
I have tried to toggle the IR_CUT, and now the colours are more accurate, but I will test it out tomorrow, it is too dark to say surely. Thank you!

roboterm, 2019-02-27:
When I was using the cameras in the Mi Home app the ir_cut never appeared as far as I remember. 
Also there is no Setting for the ir_cut in the Mi Home app Camera Settings. 
So maybe we can keep it optional always off (and also on if you like it @fliphess).

fliphess, 2019-02-27:
@roboterm: I've checked and IR Cut is used in the `ir_sample` (the original switcher), hence the clicking sound when the lights are switching on and off.... 

I remember hearing it click at night before i flashed the firmware....
It's possible IR Cut was on by default (the miihome app stream was always a bit blueish wasn't it?)