miicam/MiiCam #12: The auto_night_mode utility can be improved a lot

fliphess, 2019-02-27:
This is a collection ticket to combine all feature requests and bugs regarding the `auto_night_mode` utility. 

As I never used it myself, getting a useful and functional tool is hard for me, so I need more input of people that are actually using it. 

All input is welcome and appreciated! 👍

roboterm, 2019-02-27:
I also wrote it at #10:
When I was using the cameras in the Mi Home app the ir_cut never appeared as far as I remember. 
Also there is no Setting for the ir_cut in the Mi Home app Camera Settings. 
So maybe we can keep it optional always off (and also on if you like it @fliphess).

fliphess, 2019-02-27:
@roboterm: I've checked and IR Cut is used in the `ir_sample` (the original switcher), hence the clicking sound when the lights are switching on and off....

I remember hearing it click at night before i flashed the firmware....
It's possible IR Cut was on by default (the miihome app stream was always a bit blueish wasn't it?)

Removing it is not really an option, only if it wouldn't work (as with the auto focus), but you can leave it off by default:

- Enable save state in `config.cfg`
- remove the `-c` flag from the auto nightmode switcher commandline arguments.
- Turn the ir_cut off in the ui/mqtt or through the cli 
- save the state with the ir_cut off

After a reboot at boot time the ir_cut will shortly be enabled at boot time, but a second later the state is restored and turned off again.

fliphess, 2019-02-28:
We moved the discussion about the functionality to discord, as it's easier to determine what values and behaviour to implement while discussing rather than the slow messaging here @ github.

The current set we're working with is:
```
EV_ON_VALUE=30
EV_OFF_VALUE=1000

IR_ON_VALUE=90
IR_OFF_VALUE=100

IF EV < $EV_ON_VALUE and IR > $IR_ON_VALUE  and nightmode != "on":
    Turn on: [ir_led on;nightmode on;ir_cut off]

ELSE IF EV > $EV_OFF_VALUE and IR < $IR_OFF_VALUE and nightmode !=  "off":
    Turn off: [ir_led off; nightmode off; ir_cut on]
```

The `ir_led` and `ir_cut` are both optional, the (default) values can be changed, so only the actual black and white switch (nightmode) is always switched.

Let me know if it works for you and how it can be changed other than changing the command line arguments.

You can test by moving [the the files in the tarbal](https://fliphess.com/miicam/auto_night.tgz) in `bin/*` to `/tmp/sd/firmware/bin` and the files in `lib/*` to `/tmp/sd/firmware/lib`.

Don't use for production, only for testing ;)

fliphess, 2019-05-02:
Closing this as I havent any complaints anymore after the changes.

closed 2019-05-02 by fliphess