miicam/MiiCam #13: Openhab2 integration

manuw2009, 2019-03-18:
Hi,
I am trying to integrate this in openhab2 (openhabian) through MQTT but I can't seem to configure it properly.
Within Openhabian, I have defined an embedded broker, and assigned its IP to config.cfg.
I then created a generic MQTT thing in openhab, and created a channel to manage the brightness through home/my_camera_name/brightness.
It does even display the right value, but each time I want to change the value, it is getting back to the previous value. everything seems like the camera is pushing its value all the time,as seen in mqtt.log on the camera :+1: 
Mon Mar 18 18:32:10 UTC 2019 - Incoming MQTT message: home/camera/brightness 169
Mon Mar 18 18:32:11 UTC 2019 - Incoming MQTT message: home/camera/camera/modes {"mirror_mode":"off","flip_mode":"off","night_mode":"off","ir_cut":"on"}
Mon Mar 18 18:32:11 UTC 2019 - MQTT Action: Sending isp328 mode status
Mon Mar 18 18:32:11 UTC 2019 - Incoming MQTT message: home/camera/contrast 128
Mon Mar 18 18:32:11 UTC 2019 - Incoming MQTT message: home/camera/hue 128
Mon Mar 18 18:32:12 UTC 2019 - Incoming MQTT message: home/camera/saturation 128
Mon Mar 18 18:32:12 UTC 2019 - Incoming MQTT message: home/camera/night_mode off
Mon Mar 18 18:42:50 UTC 2019 - MQTT Update: Sending status update to mqtt broker
Mon Mar 18 18:42:50 UTC 2019 - Incoming MQTT message: home/camera/leds/blue on
Mon Mar 18 18:42:51 UTC 2019 - Incoming MQTT message: home/camera/leds/yellow off
Mon Mar 18 18:42:51 UTC 2019 - MQTT Action: Sending led status
Mon Mar 18 18:42:51 UTC 2019 - Incoming MQTT message: home/camera/leds/blue on
Has anyone successfully set up MQTT yet with this hack ?
Please help...

fliphess, 2019-03-19:
Hey @manuw2009! Thanks for creating an issue about this :) 

(I've added some back tics to your message to aquire some readability....)

I've made lots of changes to the MQTT service last months and to be very honest: I don't have any MQTT control for the MiiCam in my home automation at the moment, so most of the changes are not thoroughly tested.... 

I need some time this weekend to create some configuration for my home-assistant to test all the MQTT functionalities to test this and to verify all available topics. I think I can fix it after the weekend.

Make sure you don't overstress the camera by sending too much actions in a small amount of time, as each action takes at least 5 seconds to complete. The logs show many commands within a few seconds, It's possible at that time the camera is already overloaded at that time. Try testing manually with `mosquitto_sub` first before you add it to openhab :)

manuw2009, 2019-03-20:
Hi,
Thank you very much for your help.
Don't know if this can help, but I noticed that even
without my openhabian switched on, I am seeing the "MQTT Action: sending..." in the mqtt.log like every 2sec.

fliphess, 2019-03-21:
Thanks for your addition, that sounds like a possible loop in the mqtt logic....
I will have a look at it this weekend and get back to you about it after the weekend.

manuw2009, 2019-03-27:
Hi, did you have a chance to look into it ?
Thank you in advance !

manuw2009, 2019-04-03:
Do you want me to provide some more information maybe ?

fliphess, 2019-04-07:
Hey @manuw2009 

I'm very busy atm with some other projects that are eating all my spare time (and I think it's a good thing because I love it)...... 
Therefor I'm not as present as I was earlier. 

To find the cause of the camera not responding to MQTT, I would suggest isolate it first: Run a mosquitto docker, connect the camera to it and don't connect it to any home automation project. 

When you have a setup with only the camera connected, you can debug the camera using the mosquitto-pub utility and test what works and what not without overloading the camera. 

I've written some docs about debugging MQTT using the utilities on the camera and this is a good start to figure out if it's the camera that is not responding or the home automation that is causing the camera to overload. 

Let me know which endpoints are functional and which ones aren't that should make assisting with debugging a lot easier.

g8dm8d, 2020-03-30:
In fact, the mqtt_control.sh script listens to ${MQTT_TOPIC}/# and each "status" message received send a message on "status" topic hence the loop.
For example, a message on ${MQTT_TOPIC}/ir_cut/ calls the ir_cut_status function which send status message back on ${MQTT_TOPIC}/ir_cut/.

Postfixing "status" topics with "/get" could be a solution.

elchanly, 2020-04-25:
Hi! I cant switch on rtsp through mqtt. Is it related? I'm not so skilled at mqtt, so I'm afraid it's my fault. I send (mqtt_topuc) /rtsp/set on but nothing happens

elchanly, 2020-05-09:
hi! Don´t know too much about this, my skills are not so good, but i could make it work:
What i tried in mqtt-control.sh is this: 
`
 case "${LINE}" in
        "${MQTT_TOPIC}/rtsp/set on")
                  /tmp/sd/firmware/init/S99rtsp start
        ;;
        "${MQTT_TOPIC}/rtsp/set off")
                  /tmp/sd/firmware/init/S99rtsp stop
        ;;
        esac
`
As i'm not so skilled i cant guess why i cant use mqtt message payload, as you can see i use the whole line (using "${VALUE}" didnt work for me, i'll keep on trying)
I'll keep on working around on this for the RTSP, until @fliphess tells me "let adults work" XD