miicam/MiiCam #15: Can't boot

pimoon, 2019-04-05:
I follow the instructions but can't make my camera booting and be accessing in local : 
- If I reset the camera and boot with the sd card, the yellow led is flashing indefinitely.
- If I reset the camera and boot without the card, I can configure the wifi through Mi Home app and get fix blue led. But then when I reboot with the SD card I can't access the camera in local with it's own IP, although it is still working with the mi home app.

So I guess my camera don't look at the files on SD card.

Here is the actual firmware, should I upgrade ?
[image]

pimoon, 2019-04-05:
Well after further investigation I think my camera has a problem because it cannot detect the SD card inserted with Mi home app.
I tried with 3 differents SD card, 32Gb and 8Gb. Format in FAT32 under MacOS and Ubuntu.

pimoon, 2019-04-05:
it's finally a hardware issue.

closed 2019-04-05 by pimoon

fliphess, 2019-05-02:
@pimoon Was it a hardware issue after all or did you manage to get it to work eventually?