miicam/MiiCam #17: Loosing rtsp connexion

tarteAuxPommes, 2019-04-16:
Hi,

I'm using my camera on a domotic system called jeedom. I have to use rtsp and when i plug my camera and check on my domotic, it works. 
But after some hours (randomly), i loose the rtsp connection. I have to reboot the camera...
When i loose the rtsp, the webserver still running. 

Can you tell me why and what can i do ? I would like to avoid rebooting with cron because if somebody get in my house while rebooting...

Thanks,

fliphess, 2019-04-23:
Hey @tarteAuxPommes! Can you upload some logging with the timestamps of the time the camera stopped working? It can be a cronjob or something else that is causing this. 

I'd like to see both the boot log, the syslog and the rtspd.log to find some relations.

tarteAuxPommes, 2019-05-06:
Hi, reinstalled the hack and it works.
Thanks,

closed 2019-05-06 by tarteAuxPommes