miicam/MiiCam #19: Integration with Home Assistant platform:

shayshtern, 2019-04-20:
I don't know why but I don't success add the camera to there platform. This is the parameters in general https://www.home-assistant.io/components/generic/
If some know or what to help I will be very thankful for that.
Have a good day and thanks to everyone who what to help!
Shay Shtern.

pcmester, 2019-04-22:
Modify this, and add to your configuration.yaml file
camera:
 - platform: ffmpeg
   name: The name of your camera
   input: -rtsp_transport tcp -i rtsp://username:password@theipaddressofthecamera:554/live/ch00_0

Now you can use the new stream component, for me this is a little bit buggy, but you can give it a try by addig this to the configuration.yaml

stream:

shayshtern, 2019-04-23:
its still give me no photo.. : look: 
[image]

pcmester, 2019-04-23:
Can you play the stream in VLC by entering this (modify with your own credetials)? `rtsp://username:password@theipaddressofthecamera:554/live/ch00_0`

shayshtern, 2019-04-24:
Yes. And stream is working on VLC with this format ! With my user and password.

shayshtern, 2019-04-24:
Dose it work on your home assistant?

pcmester, 2019-04-24:
Yes, it works for me. If you are using Raspberry Pi, you can also try to disable the stream component, it can be hard to transcode a camera stream for a Raspberry. You can do that by removing the `stream:` from your  configuration.yaml file.

shayshtern, 2019-04-24:
Actually this honeassistant runs on a Linux VM on Windows 10. Is there any connection with the issue?

pcmester, 2019-04-24:
I think no, but it can be some networking issue, because of the VM. Which Linux system do you use? The Hass.io VM image, or some linux distro with Docker?

shayshtern, 2019-04-25:
Ok so let's find the issue! It's virtual box software that runs Hass.io just like on the video here: https://youtu.be/vnie-PJ87Eg 
And this is the instructions  https://www.dropbox.com/s/kwb9b3ji01leg96/Installing_HASSOS_on_VirtualBox.pdf?dl=0
It's a Linux Ubuntu 64 bit.

pcmester, 2019-04-27:
Okay, so this is the offical virtual environment image of Hass.io. It can be a networking issue, if the HA setup is in an another network. How your network looks like? Is the camera and HA on the same network?

shayshtern, 2019-04-27:
It's all on the same network. The windows 10 PC that holds the home assistant VM is on the same network as the camera.

fliphess, 2019-05-02:
Try to view the camera with `ffmpeg` instead of VLC, this should work as `ffmpeg` will play almost everything that contains some image or audio data. 

You might have to fuddle a bit to find the right command line arguments, but eventually if you can do this with ffmpeg, try using the same command line arguments in the ffmpeg component in hass. 

Otherwise if you're on windows: Check the firewall ;)

fliphess, 2019-05-02:
I don't think I can be of much help here atm as I'm not using home assistant directly with the camera anymore as the stream (and the camera) kept crashing and glitching because of too many rtsp connections at the same time. 

Home assistant opens a new connection to the camera for each viewer, and with a laptop, a regular computer and a phone all watching the camera stream at the same time it became a bit busy for the camera as it has a very tiny brain :)

I changed my setup to have zoneminder in between (zoneminder is old crap but it has a great camera proxy build in), which connects to the camera and uses the cpu power of the pc to stream to multiple viewers, is then linked with home assistant.

If you have some spare compute power i'd recommend looking at [shinobi](https://shinobi.video/) or similar.

Let me know if you find a solution, so I can add it to the documentation!

fliphess, 2019-05-02:
Okay, I've done a quick check to be sure and this works for me: 

```
camera:
  - platform: ffmpeg
    input: -rtsp_transport tcp -i rtsp://stream:bJ2xnahtCgninraelmI@camera3.home:554/live/ch00_0
```

Which results in your camera being accessible in home assistant as `camera.ffmpeg`.

shayshtern, 2019-05-02:
Only when Im on my iPhone the picture show for some seconds, that it @fliphess .
When I click its only the "loading ciricle".
camera:
 - platform: ffmpeg
   name: Xiaomi Camera
   input: -rtsp_transport tcp -i rtsp://username:password@ip:port/live/ch00_0
default_config:
panel_iframe:

fliphess, 2019-05-02:
Check your home assistant logs, it should mention some related logging as it initially works. 

Try the ffmpeg method first on the commandline before you use it in hass.
This is easier as you can quickly identify errors with the stream capture. 

If this works perfectly, but in hass it doesn't it's probably not a problem in the camera but in your hass setup, browser, firewall etc.

shayshtern, 2019-05-02:
I'm using the ffmpeg method just like this:
camera:

platform: ffmpeg
name: Xiaomi Camera
input: -rtsp_transport tcp -i rtsp://username:password@ip:port/live/ch00_0
default_config:
panel_iframe:

shayshtern, 2019-05-02:
and the logs: 
> 2019-04-24 17:26:03 WARNING (MainThread) [haffmpeg.tools] Timeout reading image.
> 2019-04-24 17:26:35 WARNING (MainThread) [haffmpeg.tools] Timeout reading image.
> 2019-04-24 17:26:40 WARNING (MainThread) [haffmpeg.tools] Timeout reading image.
> 2019-04-24 17:27:00 WARNING (MainThread) [haffmpeg.tools] Timeout reading image.

fliphess, 2019-05-02:
What errors do you see on the commandline using ffmpeg?

shayshtern, 2019-05-02:
> > 2019-04-24 17:25:12 ERROR (MainThread) [homeassistant.components.device_tracker] Unable to load /config/known_devices.yaml: Config file not found: /config/known_devices.yaml
> 
> 2019-04-24 17:25:36 ERROR (MainThread) [frontend.js.es5.201903310] http://myip:8123/frontend_es5/6113bb9c6f86bd787b06.chunk.js:2:308261 Uncaught TypeError: Cannot read property 'isOff' of undefined
> 2019-04-24 17:25:36 ERROR (MainThread) [frontend.js.es5.201903310] http://myip:8123/frontend_es5/6113bb9c6f86bd787b06.chunk.js:2:308260 TypeError: undefined is not an object (evaluating 'e.isOff')

2019-04-24 18:25:12 ERROR (MainThread) [homeassistant.components.updater] Got unexpected response: None

fliphess, 2019-05-02:
@shayshtern stop trying in home assistant now and focus on the camera: Try to connect from your server using ffmpeg on the commandline without hass in between to verify on what side the issue appears. 

If the connection issues are appearing in ffmpeg too, it's a camera issue and you can create a new issue.  If the connection issues are not there when you only use ffmpeg without hass, it's another issue not related to the camera and I can't help you solve it.

I'm closing this now, please do some proper research first and only create issues if you are very sure it's a bug in the camera and not either human error or another issue that is related to your home setup instead of the camera.

If you need help setting up, ask the friendly people on the discord channel for some advice, if you answer their questions properly they might be willing to point you in the right direction.

**Keep in mind this is not a helpdesk or community website, but a bug tracker for an open source project. The helpdesk are you :)**

closed 2019-05-02 by fliphess