miicam/MiiCam #20: Camera cannot connect to hidden SSID

simoncirstoiu, 2019-04-30:
I've tried few times, and the camera cannot connect to hidden WIFI SSID

Is there a setting that must be modified in order to be able to connect to a hidden SSID?
I edited correctly the name and the password, and if is not hidden, everything works ok, but when the ssid is hidden the camera refuse to connect anymore.

fliphess, 2019-05-02:
Hey @simoncirstoiu 

For wifi configuration we still fully rely on the utilities in the original firmware, as all we do is save the wifi credentials to nvram, connecting is done using `wpa_supplicant` and some scripts hidden somewhere in the firmware image. 

I've found [some topics on SO](https://unix.stackexchange.com/questions/257284/failure-to-connect-to-hidden-ssid-with-wpa-supplicant-on-debian) that seem to be related and if this is the case it's due to a missing configuration setting. 

I don't think I can easily create a setting in config.cfg to change the `wpa_supplicant` configuration but I'm reluctant to say it's not possible. 

What might work as a  workaround: Copy the `wpa_supplicant.conf` from the etc directory in memory to `/tmp/sd/firmware/etc/wpa_supplicant/` and as a result it will be copied in place every time at boot time. 

I'm not sure sure if it works, but it's worth a try :)

fliphess, 2019-05-02:
Found it! In `/mnt/data/ot_wifi_tool/wifi_start.sh` on the camera you can change the wpa settings as this is where the `/tmp/wpa_supplicant.conf` is generated at boot time. 

You can change the settings to support a hidden SSID but this is an "advanced topic" (in other words, you are on your own)

Be aware that you are changing files in the writable storage on the camera. (As in the files I've warned not to change in several docs and in the security considerations) This is reversible when using SSH, but if your camera doesn't get online after the config change, consider your camera bricked.  Having to connect a serial port to the camera is not very easy so **be careful**.

fliphess, 2019-05-02:
I've added this issue to [the documentation](https://miicam.github.io/Troubleshooting/#wifi-connection-issues) as a limitation of the camera, so closing this. 

Btw, I add this functionality to the list of wannahaves as I prefer to have more control over the boot process and wifi setup in general, but for now I'm sorry to say: "Wontfix"...

closed 2019-05-02 by fliphess