miicam/MiiCam #21: Motion sensibility

tarteAuxPommes, 2019-05-06:
Hey,

I would like to know if it's possible to add a parameter for the motion sensibility ? It looks very sensitiv by default and i'm receiving notification constantly.

Thanks for you work,

fliphess, 2019-05-06:
Hey @tarteAuxPommes  I think it's possible, but you've to dive deep in the SDK. 

Have a look at the samples in the toolchain for some examples:
- https://fliphess.com/toolchain/Software/Embedded_Linux/source/gm_graph/gm_lib/samples/encode_with_capture_motion_detection.c 
and 
- https://fliphess.com/toolchain/Software/Embedded_Linux/source/gm_graph/gm_lib/samples/encode_with_capture_motion_detection2.c

There's a lot more features available that are not present in the rtsp daemon yet....

tarteAuxPommes, 2019-05-06:
Hi, Thanks for your reply. Unfortunately i'm afraid that i can do nothing at my level with this.. :/

fliphess, 2019-05-06:
Me neither :) But It's on my list to implement as soon as I have a bit more experience in C :)
We'll get there, it's just a matter of time :)