miicam/MiiCam #23: Help with my camera

HITMIK, 2019-06-09:
(sorry for my english)
I was install this manual:

format SD card to FAT32
copy tf_recovery.img to card
plug sd card to camera and power on
blue led is blue
format SD card to FAT32
copy sdcard folder to card
plug sd card to camera and power on
blue led is blue
Dont work http://my_ip_address

Where is problem?

closed 2019-06-09 by support[bot]

HITMIK, 2019-06-09:
@fliphess

fliphess, 2019-07-05:
Hey @HITMIK ! Did you manage to get it working? 

Without much information, the issue you describe sounds like a configuration error with which the guys on discord can help you if you ask nicely :) 

I'm not working on the camera much at the moment due to busy with work and because of the summer...... If you need some help setting up the camera have a look at the [discord server](https://discord.gg/rjW2SSB) for some assistance setting up.