miicam/MiiCam #25: Camera setting after reboot

glelan, 2019-08-05:
Hello It look like all camera settings like "Auto Sharpen" "Auto Expo" etc... are allways "activated" after reboot even if it was desactivated before reboot and despite what value we put in config editor / RESTORE_STATE
Have try with RESTORE_STATE="0" or RESTORE_STATE="1"
no change it allways come back at "on" with thoses options.
an idea?
Thanks

fliphess, 2019-10-28:
Hey @glelan (Sorry for the late response and the stalebot) 

What you say is correct, there is no way (yet)  to set the Auto Exposure, Noise reduction etc (The top buttons in the UI) at boot-time, as there is no state saved for these options.

The default value that is set when loading the kernel module that takes care of these settings are always set at boot time and never override in the restore_state script. 

I will try to add them in the not-so-near future as it would be nice to have :)

fliphess, 2020-01-02:
I've removed the mentioned settings from the frontend too as their automagically set by the firmware at boot time and I'm not sure the settings add any value as I've never felt the use to change the default setting.

closed 2020-01-02 by fliphess