miicam/MiiCam #28: Support for CMSXJ16A ?

1300371, 2019-12-28:
Describe in the issue you are experiencing, what behaviour you see, if you see any errors and most important if something is not working: send the log files, I really need to see them to help you debug.

...

I'm trying to use this hack with this camera : banggood.com/International-Version-Xiaomi-Mijia-IMILAB-Xiaobai-H_265-1080P-Smart-Home-IP-Camera-360-PTZ-AI-Detection-WIFI-Security-Monitor-from-Xiaomi-Eco-system-p-1450083.html

When putting the hack in SD, i've got logs.
**Jan  1 01:00:09 kernel: ERROR!!! MT7601 E2PROM: WRONG VERSION 0xd, should be 9**
````
Jan  1 01:00:09 kernel: Key2Str is Invalid key length(0) or Type(0)
Jan  1 01:00:09 kernel: Key3Str is Invalid key length(0) or Type(0)
Jan  1 01:00:09 kernel: Key4Str is Invalid key length(0) or Type(0)
Jan  1 01:00:09 kernel: ###### Force at HT20 (BW_20) mode !!! ########
Jan  1 01:00:09 miio_client_helper: Got _internal.req_wifi_conf_status
Jan  1 01:00:09 kernel: 1. Phy Mode = 14
Jan  1 01:00:09 kernel: 2. Phy Mode = 14
Jan  1 01:00:09 kernel: NVM is Efuse and its size =1d[1e0-1fc] 
Jan  1 01:00:09 miio_ota[209]: OTD libmortox sockfd: 4
Jan  1 01:00:09 kernel: ERROR!!! MT7601 E2PROM: WRONG VERSION 0xd, should be 9
Jan  1 01:00:09 miio_agent: [19700101 01:00:09] [INFO] OT agent listen accept sockfd: 7
Jan  1 01:00:09 miio_agent: [19700101 01:00:09] [INFO] register key: miIO.ota, fd: 7
Jan  1 01:00:09 miio_ota[209]: OTD sockfd: 5
Jan  1 01:00:09 miio_client_helper: {"method":"_internal.res_wifi_conf_status","params":1}
Jan  1 01:00:10 kernel: 3. Phy Mode = 14
Jan  1 01:00:10 kernel: AntCfgInit: primary/secondary ant 0/1
Jan  1 01:00:10 miio_algo[197]: [miio_ipc_server_connect]connect error, errno=111^M
````

I really don't know if my cam is not compatible or i'm doing something wrong.
I think yes based on this comment https://github.com/Filipowicz251/mijia-1080P-hacks/issues/53#issuecomment-418193602

Sorry if not ... and btw ty for your job

1300371, 2019-12-28:
Sorry if hack not compatible : i was thinking that is a feature request ...

fliphess, 2019-12-30:
@1300371 I think the model you are using is using a newer eeprom chip than the one in the chuangmi camera, but we might be able to make it compatible by replacing the eeprom utils in the original firmware with some code that is able to detect which chip it is and read any of them. I have some examples in the toolchain that look usable. It's a lot of work and I'm not sure if i've enough knowledge to do it, but it's worthwhile researching :) 

I'll keep this issue open for a bit so I can have a look what is possible (without making any promises over here).....

1300371, 2020-01-03:
@fliphess Sure, thanks for the effort !  
Let me know if you need more info :+1:

OUARZA, 2020-01-11:
Hello,
I am also interested. I have a CMSXJ16A camera.
Cordially.

eduardoaly, 2020-01-13:
Hello,
I am also interested too.

lapinpt, 2020-01-21:
hi, one more vote for CMSXJ16A chuangmi ipc016

fliphess, 2020-02-12:
TLDR; No.

I had a look at the firmware used for the `CMSXJ16A` or ipc016, and I don't think I will be able to adjust this firmware to support it as it's a completely different setup. 
The internal hardware appears to be quite different and the firmware image is setup by another group in another way. (A cleaner, more sophisticated way, but different enough for not easily being able to make this hack compatible with it)

I'm sure eventually there will be some crazy geeks that will take the time to find a hack and create some nice set of tools to work with the ipc016  camera, and If someone sends me one for testing and one for desoldering I might be interested in assisting these other crazy nerds as well, but supporting multiple crappy Chinese low budget baby camera's is not exactly what I had in mind with my spare time after work for now as it's already getting harder to find the motivation to work on this single one. 

For now I will close this issue as there are no future plans to support this camera.

closed 2020-02-12 by fliphess