miicam/MiiCam #29: About RTSP encryption

array81, 2020-01-02:
Sorry for this message but I cannot find a email or other way to contact the author.
About "WARNING: The RTSP daemon has authentication but without encryption! Read the Security Considerations for more information" on your guide, if I use the firmware to enabled RTSP and connect the cam with my Synology NAS (on the same network) can I have security problem?
I think the "encryption" is a big problem on internet but a small problem to connect cam and nas, is't right?

closed 2020-01-02 by support[bot]

fliphess, 2020-01-02:
Hey @array81 :) 

I'm pretty sure more security minded people will crucify me for saying this but I think you are right. 

The camera is not in any way made for being used as a serious security device used for critical surveillance so If you are using this camera (for example) for monitoring a bank vault , I think you are doing it wrong. I use this camera at home for detecting if my cat is on the other side of the door and imho this is not too much of an issue.

The camera does not use encryption on the RTSP stream, so technically it is possible to capture the stream password (and thus the stream itself) using tcpdump as soon as you have joined the same wifi network. 

For me this is not an issue, I use it at home and everyone who is on my WIFI are trusted enough to watch my RTSP stream while in (or close by) my house, as all you can see is the other side of the door, which I don't stand naked in front of :)

I think you should decide for yourself if it's secure enough, just as you should decide for yourself if using any other service that captures your data (either online or offline) is secure enough, as did I and for me it's not even the most insecure device in my network....

fliphess, 2020-01-02:
BTW if you can separate things with a dedicated IOT wifi SSID and some firewall rules to prevent regular users from visiting iot wifi ip space, that be even better ;)