miicam/MiiCam #30: RSTP not working on MJSXJ01CM

SuperSpe, 2020-01-25:
[ ] - Issue or bug is not in: https://github.com/fliphess/chuangmi-720p-hack/wiki/Troubleshooting)
[ ] - I've attached a log file (https://github.com/fliphess/chuangmi-720p-hack/wiki/How-to-tar-or-zip-the-logs-when-filing-an-issue)

I can acces to all hack functions like http, sshd etc, but cannot get the rtsp stream working.
I try to disable the authentication, no results.

I try ffprobe rtsp://localhost:554/live/ch00_0
and got
[tcp @ 0x2e7c0] Connection to tcp://localhost:554?timeout=0 failed: Connection refused
rtsp://localhost:554/live/ch00_0: Connection refused

any advice?

**Describe in the issue you are experiencing, what behaviour you see, if you see any errors and most important if something is not working: send the log files, I really need to see them to help you debug.**
...

SuperSpe, 2020-01-25:
fuck

fliphess, 2020-01-26:
@SuperSpe Just from the information that you provide it looks like the RTSP server cannot be started. There can be several causes, but I suspect the xiaomi cloud streaming service being running causing the rstp not being able to open the required sockets and sources. 

can you check in the logs if there's any error at startup time?

fliphess, 2020-01-26:
if you check the process list (`ps -ef | grep rtsp`) can you see a running process? If not, try starting it manually (`/etc/init/S99rtspd start`) and check the logs what went wrong?

SuperSpe, 2020-01-27:
Yes I see the running process. I can restart the processo too with the same result.
The cloud streaming service is disabled in the config btw.
I tried severl version (old) of the hack, no luck
This is what I got if I run the rtsp manually:

root@camera /root # /tmp/sd/firmware/bin/rtspd -f15 -w1280 -h720 -b4096 -m4
22:32:17 INFO  rtspd: Starting the RTSP Daemon
Error! Abnormal AP exit under chipid(0),cpuid(1),graph(1),query_id=0x10001
22:32:17 INFO  rtspd: Width        : 1280
22:32:17 INFO  rtspd: Height       : 720
22:32:17 INFO  rtspd: Encoder      : 184760
22:32:17 INFO  rtspd: Framerate    : 15
22:32:17 INFO  rtspd: Bitrate      : 4096
22:32:17 INFO  rtspd: Bitrate Mode : 4
22:32:19 ERROR rtspd: GM Poll timeout

I am getting the last gm pool timeout, like every few seconds

fliphess, 2020-01-27:
```
22:32:19 ERROR rtspd: GM Poll timeout
```

GM Poll timeout indicates it has something to do with the kernel module that is used for capturing frames from the sensor (GMLib is the library created by the provider to do this and the poll timeout comes from this library) 
Can you provide the output of the `dmesg` command? 

Chances are it will never work, as the camera is a completely different model, but afaik they're using the the same board to run the camera. If you want pan and tilt there are some challenges on your way, as the chuangmi camera itself doesn't provide pan and tilt.

SuperSpe, 2020-01-28:
Hi. I am sad :(

dmesg is full of this:

vcap_lli_irq_handler: 330 callbacks suppressed
[VCAP_ERR]: [0] VI1 no clock!
[VCAP_ERR]: [0] VI1 no clock!
[VCAP_ERR]: [0] VI1 no clock!
[VCAP_ERR]: [0] VI1 no clock!
[VCAP_ERR]: [0] ch#4 timeout!(T:112ms)
[VCAP_ERR]: [0] VI1 no clock!
[VCAP_ERR]: [0] VI1 no clock!
[VCAP_ERR]: [0] VI1 no clock!
[VCAP_ERR]: [0] VI1 no clock!
[VCAP_ERR]: [0] VI1 no clock!

SuperSpe, 2020-02-01:
So like I was saying on discord, I fix the problem by using another firmware version and not the one on the repo.
The firmware for this board can be found here
https://1drv.ms/u/s!Ar5C-aX9A4YAkBRlGPgJKocdqVRw

fliphess, 2020-02-03:
Good to hear you got it to work :) 

 I checked the md5sum of the url you posted and the tf_recovery.img file in the repo, and they're exactly the same: 
```
MD5 (tf_recovery.img) = bf2705825ba943bb1dff2541ce2afef2
MD5 (tf_recovery.img1) = bf2705825ba943bb1dff2541ce2afef2
```

So it must be either magic or skill that made it work :) (Or a bit of both)

closed 2020-02-03 by fliphess