miicam/MiiCam #32: Auto night mode configuration

hiol83, 2020-01-26:
There is a difference between the documentation about auto night mode and the default configuration in the firmware

Actual default configuration in the last release v0.98.2
AUTO_NIGHT_MODE_ARGS="--ev-on 30 --ir-on 90 --ev-off 1000 --ir-off 100 --delay 5 --set-ircut --set-irled"
Dafault configuration in the documentation 
AUTO_NIGHT_MODE_ARGS="-d 120 -e 200 -i 5 -l -n -v"

Actually, i would like to know what these commands mean (--ev-on 30 --ir-on 90 --ev-off 1000 --ir-off 100 --delay 5 --set-ircut --set-irled) because the ir is switching on correctly when is dark but then it always stay on.

roboterm, 2020-01-26:
These commands configure the behavior of the Auto Night mode. The best practice is to find out the wanted values for yourself by running the Auto Night mode in a console window where the program can output all variable values.  
https://miicam.github.io/configuration/Using-Auto-Night-Mode/

roboterm, 2020-01-26:
For me this configuration works;
AUTO_NIGHT_MODE_ARGS="--ev-on 700 --ir-on 100 --ev-off 2140 --ir-off 130 --delay 5 --set-irled --set-ircut"

hiol83, 2020-01-27:
thank you @roboterm. uploaded your configuration and it's working fine also for me. 
i was also looking about documentation. but these commands are not well explained there. i'm not used with linux command line or similar. if i copy and paste example in documentation auto night mode simply doesn't work. that's why i thought about suggestin an issue.

anyway again, about my issue, seem that you helped me a lot. so, many thanks to you

roboterm, 2020-01-27:
you're welcome!

closed 2020-02-03 by fliphess