miicam/MiiCam #36: Commande ligne for shutdown

chris777c, 2020-05-02:
Hello,

I search on system chuangmi how to shutdown properly because I stop camera with smart plug and would like shutdown system before to conserve my sd card longevity ?

I find poweroff (in /sbin/ folder) command but it's fake because don't turn off camera only reboot.

Best regards.

fliphess, 2020-05-02:
Hey @chris777c , I've tried several ways of fully shutting down the camera or implement a hibernation or sleep mode but other then disconnecting the power plug I don't think it's possible.

There is no sophisticated power management mechanism build in as you find in a computer or in newer electronics and as such it's not possible power down the electronics... I think this is the way the hardware was designed by the board creators.  

(Even in the original firmware the "sleep" mode is not much more than shutting down the video stream and turning off the leds)

If you really want a smart power on and off functionality for the camera you might want to accomplish this by relay switching the power outlet.

chris777c, 2020-05-02:
Ok thank you for spend your time to reply me.

closed 2020-05-02 by chris777c