miicam/MiiCam #37: I can't access the camera by ip address

juanvivo, 2020-05-03:
I have followed the tutorial to try to modify the camera and be able to use it with the functionalities that you indicate.

I signed it with the tf_recovery.img file and the process seems to work fine. On the card is the .bak file and a couple of folders that are created. I connect the camera to the wifi with the original "my home" app. Then I downloaded and copied the contents of the "sdcard" folder to the memory card, modified the wifi name and password settings, and entered the sd in the camera.

The camera starts and after a few seconds the blue led remains solid.

But if I try to enter the IP address through the browser, it doesn't load any page.

Can you tell me where I could be failing?

I don't know where I can find the log files to attach them to ...

roboterm, 2020-05-03:
Can you connect via SSH or Telnet to the camera?
Is it visible in the Mi Home app (online-/offline-status)?
If it’s still online in the Mi Home app the hack didn’t work. 
Did you build the binaries by yourself or did you just copy the files from the zip you’ve downloaded?

juanvivo, 2020-05-04:
I am not familiar with these methods (SSH or telnet) to connect to the webcam. I will investigate to see how I should do it.

Yes, in the application of my home it works perfectly.

Don't create binaries. Use the files I downloaded. First of the last release, and in case there were problems with it, then I replaced them with those of version 0.98.2, but it did not work either.

To do the process, I followed the tutorial on this website: [TUTO](https://domoticaencasa.es/home-assistant-52-usamos-la-camara-xiaomi-chuangmi/) .I'm going to try again taking the files directly from the github.

When the instructions say: Use the MiiHome app to configure the WIFI for the first time ...
Are you referring to the official Xiaomi Home app? or do I have to install a different app?

roboterm, 2020-05-04:
Sorry for confusing the apps.
Of course I mean the official Xiaomi Home app:
https://apps.apple.com/de/app/xiaomi-home-xiaomi-smarthome/id957323480

If the camera is still visible on this app, the hack didn't work.
First try to flash the firmware with this one:
https://github.com/miicam/MiiCam/blob/master/tf_recovery.img

You can find instructions how to do that here:
https://miicam.github.io/Flashing-the-U-boot-firmware-to-an-older-version-when-the-hack-is-not-working/

Configuring the camera's wifi access via the Xiaomi Home app is recommended to get the ip adress the camera has. You need it later.

I recommend you to follow the Getting Started here (not the short version ;) ):
https://miicam.github.io/getting-started/

juanvivo, 2020-05-04:
thanks for the quick response

the hack still doesn't work. Could it be that the problem is that my camera is not the global version?

I have gone through all the steps that you indicate and the result is the same, I cannot access it by putting the local ip in the browser.

The process is:

- record the image
- configure the wifi with the app my home. (I have tried not to do this step, and then look at the ip on the router, but the webcam does not connect to the wifi, the flashing orange LED always stays)
- copy the files from the sdcard folder to the root of the sd

it is right?

roboterm, 2020-05-04:
How to flash the firmware

To revert this behaviour, you can flash the firmware of the camera to an older u-boot version that still allows booting from SD.

To activate this firmware, use the tf_recovery.img file:

Place the tf_recovery.img in the root of the SD card and remove any other files.
Slide the SD card in the slot and power up the camera.
Wait for 10 minutes till the blue led goes on.
Unplug the camera’s power.
If the flash is successful, the tf_recovery.img file is renamed to tf_recovery.img.bak.

That’s copied from the documentation I referred. Just follow the link and read the instructions. 

- other than the documentation says you can enter the WiFi credentials into the config file before you insert the sd card into the camera. 
- I don’t know how the hack works if you just copy the files from the downloaded content to the sd card. I built the binaries by myself. This is also noted in the documentation.

juanvivo, 2020-05-05:
Don't ask me how, but it works.

I have not done anything different, I have repeated the process once more pointing everything I was doing step by step as a guide for dummies and now it works.

I have thought about publishing the step by step guide for clumsy people like me, and even recording the tutorial on YouTube, but I think I have read that you did not want things to be published on the subject. If you want the guide with the step-by-step notes I have made, tell me.

thanks for your help.

roboterm, 2020-05-05:
Congrats!

If you want to publish something on YouTube etc. you must ask @fliphess .
I'm not the owner of the repo - just trying to help :)

juanvivo, 2020-05-05:
tks @roboterm 

I'll wait for @fliphess to tell me something before posting anything ...

fliphess, 2020-05-08:
Hey @juanvivo! If you want to create a youtube video of how to install the hack, that'd be great :) Let me know where to find it so I can link to it from the docs 👍