miicam/MiiCam #38: Night mode is not activated

juanvivo, 2020-05-05:
Finally I got the camera to work and it's already mounted and broadcasting. :)

I am having difficulty activating night mode. Although I modify the variable AUTO_NIGHT_MODE = 1, the night mode is not activated, and I continue taking low quality color images ...

The line that I have in the configuration is the one that comes by default:

`## Command line arguments for the auto night mode and irled switcher
AUTO_NIGHT_MODE_ARGS = "- ev-on 30 --ir-on 90 --ev-off 1000 --ir-off 100 --delay 5 --set-ircut --set-irled" `

Maybe I have to modify any of these parameters to activate the night mode? In the configuration before the modification of the camera the night mode entered the same conditions without problem.

roboterm, 2020-05-05:
For me this configuration works;
`AUTO_NIGHT_MODE_ARGS="--ev-on 700 --ir-on 100 --ev-off 2140 --ir-off 130 --delay 5 --set-irled --set-ircut"`

juanvivo, 2020-05-05:
tks again @roboterm . I'll try now to see how it goes tonight.